adafruit/adafruit-beaglebone-io-python #344: SPI not working in BeagleBone Black

caxefaizan, 2020-11-13:
Hi. i am using BBB to get values in SPI mode but am not able to find a proper documentation to make it work. I have connected the pins for SPI0 with the Ic from which i want to get the values.
so my code includes
`from Adafruit_BBIO.SPI import SPI`

then i initialize the spi as` spi = SPI(1,0)`
followed by
`spi.open(1,0)`
`spi.writebytes([address])`

since it was not working i tried the tutorial on github.
but using `print(spi.xfer2([32, 11, 110, 22, 220])) `gives me values `[0,0,0,0,0] `also doesnt seem to work.

i used `config-pin `to setup SPI pins and also tried editing the` uEnv.txt to` include `spidev1.0`

caxefaizan, 2020-11-13:
> Hi. i am using BBB to get values in SPI mode but am not able to find a proper documentation to make it work. I have connected the pins for SPI0 with the Ic from which i want to get the values.
> so my [...]

The version.sh is
```
git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLT00C04319BBBK066A]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Stretch imgtec Image 2020-04-06]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[l                                                                             ocation: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
kernel:[4.14.108-ti-r131]
nodejs:[v6.17.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-                                                                             00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.0-0rcnee0~stretch+20200403]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~stretch+20200322]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev us                                                                             ers systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi iio                                                                              docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/m                                                                             mcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656                                                                              rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.031417] remoteproc remoteproc0: wkup_m3 is available
[    1.115417] remoteproc remoteproc0: powering up wkup_m3
[    1.115535] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf,                                                                              size 217168
[    1.118106] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   77.519415] remoteproc remoteproc1: 4a334000.pru is available
[   77.542226] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   76.592501] pruss 4a300000.pruss: creating PRU cores and other child platform                                                                              devices
[   77.519415] remoteproc remoteproc1: 4a334000.pru is available
[   77.519574] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a32600                                                                             4/pruss@0/pru@34000 probed successfully
[   77.542226] remoteproc remoteproc2: 4a338000.pru is available
[   77.542374] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a32600                                                                             4/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.673589] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.686514] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n                                                                              Wireless Network Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

pdp7, 2020-11-20:
Thanks for reporting, I will try to reproduce and investigate

dayrenefrometa, 2020-12-14:
Hi everyone.

Any update on this issue? I am facing the same problem: SPI is not working in BeagleBone Black. 

I have followed [this](https://arcanesciencelab.wordpress.com/2020/01/14/the-correct-way-to-enable-spi-ports-on-the-beaglebone-black/) to properly setup the SPI pins, but using print(spi.xfer2([32, 11, 110, 22, 220])) gives me values [255,255,255,255,255]

Thank you very much,
Regards

The version.sh is:
[spi_not_working.txt](https://github.com/adafruit/adafruit-beaglebone-io-python/files/5691911/spi_not_working.txt)

dayrenefrometa, 2020-12-14:
The version.sh is

git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLT00C02319BBBK0590]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
kernel:[4.19.94-ti-r42]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20201119.0-0~buster+20201123]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=UUID=1601b79d-e5ae-40d6-a294-cd66809d6241 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   69.627417] remoteproc remoteproc0: wkup_m3 is available
[   69.675916] remoteproc remoteproc0: powering up wkup_m3
[   69.675948] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   69.676232] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   71.965958] remoteproc remoteproc1: 4a334000.pru is available
[   71.984507] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   71.965958] remoteproc remoteproc1: 4a334000.pru is available
[   71.966164] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   71.984507] remoteproc remoteproc2: 4a338000.pru is available
[   71.984666] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.949852] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.963254] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

caxefaizan, 2020-12-15:
> Hi everyone.
>
> Any update on this issue? I am facing the same problem: SPI is not working in BeagleBone Black.
>
> I have followed [...]

No updates what so ever... I dropped working on it as there is so many differences in the documentation.

dayrenefrometa, 2020-12-17:
> No updates what so ever... I dropped working on it as there is so many differences in the documentation.

Ok, Thanks. Could you recommend another library? Did you find a similar one, for Python or C++?

Thanks in advance,

Regards.

caxefaizan, 2020-12-17:
> Ok, Thanks. Could you recommend another library? Did you find a similar one, for Python or C++?
>
> Thanks in advance,
>
> Regards.

i'm afraid cant. like i said i dropped working on it because of the lack of support and documentation

dayrenefrometa, 2020-12-18:
> i'm afraid cant. like i said i dropped working on it because of the lack of support and documentation

Hi caxefaizan,

I just wanted to tell you that although print(spi.xfer2([32, 11, 110, 22, 220])) gives me values [255,255,255,255,255], the SPI interface is apparently working well. I tried sending data using spi.writebytes([0x61]) and analyzing the outgoing signals (CLK, CS, D0, and D1) in the oscilloscope to see what I got. I have attached a picture with the transmission in action on the oscilloscope (blue-> chip select signal, green -> clock signal, pink-> data sent (0x61), yellow -> received data, 
[image]
I am not reading anything)

Best regards,

Dayrene

pdp7, 2021-01-25:
I'm testing with the [Buster IoT Image 2021-01-11 image](https://rcn-ee.net/rootfs/bb.org/testing/2021-01-11/buster-iot/):
```
debian@beaglebone:~$ grep -v ^# /boot/uEnv.txt 
uname_r=4.19.94-ti-r57
enable_uboot_overlays=1
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1
uboot_overlay_pru=AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
enable_uboot_cape_universal=1
cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet
```
Run `set_spi_pins.sh` to make sure the pins are set to spi mode:
```sh
#!/bin/bash

echo set SPI0 pins
config-pin p9.17 spi_cs
config-pin p9.21 spi
config-pin p9.18 spi
config-pin p9.22 spi_sclk

echo set SPI1 pins
config-pin p9.20 spi_cs
config-pin p9.28 spi_cs
config-pin p9.19 spi_cs
config-pin p9.42 spi_cs
config-pin p9.29 spi
config-pin p9.30 spi
config-pin p9.31 spi_sclk
```
```
debian@beaglebone:~$ ./set_spi_pins.sh 
set SPI0 pins
Current mode for P9_17 is:     spi_cs
Current mode for P9_21 is:     spi
Current mode for P9_18 is:     spi
Current mode for P9_22 is:     spi_sclk
set SPI1 pins
Current mode for P9_20 is:     spi_cs
Current mode for P9_28 is:     spi_cs
Current mode for P9_19 is:     spi_cs
Current mode for P9_42 is:     spi_cs
Current mode for P9_29 is:     spi
Current mode for P9_30 is:     spi
Current mode for P9_31 is:     spi_sclk
```
Here are the spidev device files that exist:
```
debian@beaglebone:~$ ls -ltar /dev/spidev*
crw-rw---- 1 root spi 153, 0 Jan 15 06:17 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Jan 15 06:17 /dev/spidev0.1
crw-rw---- 1 root spi 153, 3 Jan 15 06:17 /dev/spidev1.1
crw-rw---- 1 root spi 153, 2 Jan 15 06:17 /dev/spidev1.0
```
I run `test_spi.py` which is:
```py
debian@beaglebone:~$ cat test_spi.py 
from Adafruit_BBIO.SPI import SPI
#spi = SPI(bus, device) #/dev/spidev<bus>.<device>

# /dev/spidev0.0
spi = SPI(0,0)
print(spi.xfer2([32, 11, 110, 22, 220]))
spi.close() 

# /dev/spidev0.1
spi = SPI(0,1)
print(spi.xfer2([32, 11, 110, 22, 220]))
spi.close() 

# /dev/spidev1.0
spi = SPI(1,0)
print(spi.xfer2([32, 11, 110, 22, 220]))
spi.close() 

# /dev/spidev1.1
spi = SPI(1,1)
print(spi.xfer2([32, 11, 110, 22, 220]))
spi.close() 
```
The output:
```
debian@beaglebone:~$ python3 test_spi.py 
[255, 255, 255, 255, 255]
[255, 255, 255, 255, 255]
[255, 255, 255, 255, 255]
[255, 255, 255, 255, 255]
```
so this does seem to indicate a problem as this is supposed to be a loopback test.  I'll investigate further.

pdp7, 2021-01-25:
It seems the issue may not be with Adafruit_BBIO.  Here is a loopback test with the generic python spidev module which incorrectly reads `0xFF`:
```
debian@beaglebone:~$ cat spi.py 
#!/usr/bin/env python

# loopback test script
#   connect MOSI and MISO

import spidev
import time

spi = spidev.SpiDev()
spi.open(0, 0)

def BytesToHex(Bytes):
    return ''.join(["0x%02X " % x for x in Bytes]).strip()

try:
    while True:
        resp = spi.xfer2([0x01, 0x02])
        print(BytesToHex(resp))
        time.sleep(1)

except KeyboardInterrupt:
    spi.close()
```
```
debian@beaglebone:~$ python3 spi.py
0xFF 0xFF
0xFF 0xFF
0xFF 0xFF
0xFF 0xFF
0xFF 0xFF
0xFF 0xFF
```
I get the same seemingly incorrect results for the spidev test from Derek Molloy's book:
https://github.com/derekmolloy/exploringBB/tree/version2/chp08/spi/spidev_test
```
debian@beaglebone:~/exploringBB/chp08/spi/spidev_test$ ./spidev_test 
spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)

FF FF FF FF FF FF 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
FF FF 
```
@jadonk @RobertCNelson have you noticed any spi problems on newer images?

pdp7, 2021-01-25:
I've talked to @RobertCNelson and it looks like there might have been a regression in SPI since the last official image release ([AM3358 Debian 10.3 2020-04-06 4GB SD IoT](https://beagleboard.org/latest-images)).  It's possible the issue could have occurred between 4.14 and 4.19 kernel.

pdp7, 2021-01-26:
I've been advised by @RobertCNelson to try earlier 4.19 kernel builds going back to the April 2020 build.  They are all in our debian repo so it is matter of apt-get and then rebooting and retesting.   I'll update when I am able to complete that progress.

An alternative solution is to move ahead to the 5.8 kernel build in the beagle repo via apt-get.  It has been confirmed by @MarkAYoder that SPI is working ok with that kernel.