adafruit/adafruit-beaglebone-io-python #347: I2C issue with Adafruit-BBIO 1.2.0

tomek242, 2021-02-15:
Howdy good folks.
My issue is I want to use simple program for my MotorBridge Cape v1.0

import MotorBridge
import time
ServoName   =  1
Frequency   =  50
Angle1      =  45
Angle2      =  90
Angle3      =  135
Angle4      =  170
Angle5      =  135
Angle6      =  90
Angle7      =  45
Angle8      =  15 
Angle0      =  90 

if __name__=="__main__":
    motor = MotorBridge.MotorBridgeCape()
    motor.ServoInit(ServoName,Frequency)
 
    while True:
        motor.ServoMoveAngle(ServoName,Angle1)
        time.sleep(0.5)
        motor.ServoMoveAngle(ServoName,Angle2)
        time.sleep(0.5)
        motor.ServoMoveAngle(ServoName,Angle3)
        time.sleep(0.5)
        motor.ServoMoveAngle(ServoName,Angle4)
        time.sleep(0.5)
        motor.ServoMoveAngle(ServoName,Angle5)
        time.sleep(0.5)
        motor.ServoMoveAngle(ServoName,Angle6)
        time.sleep(0.5)
        motor.ServoMoveAngle(ServoName,Angle7)
        time.sleep(0.5)
        motor.ServoMoveAngle(ServoName,Angle8)
        time.sleep(0.5)
        motor.ServoMoveAngle(ServoName,Angle0)
        time.sleep(2)
 
and when i want to fired up this simple servo test i had this:

Traceback (most recent call last):
  File "/var/lib/cloud9/my_python/servotest.py", line 18, in <module>
    motor.ServoInit(ServoName,Frequency)
  File "/var/lib/cloud9/my_python/MotorBridge.py", line 345, in ServoInit
    WriteHalfWord(SVM1_FREQ,Frequency)
  File "/var/lib/cloud9/my_python/MotorBridge.py", line 212, in WriteHalfWord
    MotorBridge.writeList(WriteMode,data)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_GPIO-1.0.4-py3.7.egg/Adafruit_GPIO/I2C.py", line 129, in writeList
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO-1.1.8-py3.7.egg/Adafruit_PureIO/smbus.py", line 364, in write_i2c_block_data
OSError: [Errno 121] Remote I/O error

sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[029041f6866049997bbfd2c7667b3c6e8c95201c]
eeprom:[A335BNLT00C05114BBBK1322]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster LXQt Image 2020-05-18]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g31a8ae0206]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g31a8ae0206]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
UBOOT: Loaded Overlay:[BB-UART2-00A0]
kernel:[4.19.94-ti-r43]
nodejs:[v10.19.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-UART2-00A0.dtbo]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200430.0-0~buster+20200430]
pkg:[bb-wl18xx-firmware]:[1.20200420.1-0~buster+20200424]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   85.845243] remoteproc remoteproc0: wkup_m3 is available
[   86.427759] remoteproc remoteproc0: powering up wkup_m3
[   86.427794] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   86.428131] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   93.265874] remoteproc remoteproc1: 4a334000.pru is available
[   93.279707] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   93.265874] remoteproc remoteproc1: 4a334000.pru is available
[   93.266027] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   93.279707] remoteproc remoteproc2: 4a338000.pru is available
[   93.280000] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    1.177283] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.190496] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 005: ID 079b:0062 Sagem XG-76NA 802.11bg
Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 003: ID 0c45:6366 Microdia
Bus 001 Device 002: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

uname -a
Linux beaglebone 4.19.94-ti-r43 #1buster SMP PREEMPT Wed Apr 22 06:11:16 UTC 2020 armv7l GNU/Linux

I had this problem with any debian on my BBB...Can someon help me or someone had similiar problem? 
Thanks for any help.
Tom.

pdp7, 2021-02-15:
I am not familiar with MotorBridge.  Could you link to where that library lives?

tomek242, 2021-02-15:
Hello Drew.
Thanks for answer me.
The library for this cape it's very simple to use and its based on Adafruit-GPIO library. Below i write how to use this cape:
```
import MotorBridge
motor = MotorBridge.MotorBridgeCape()
```

And below we have installing dependiences:

`sudo apt-get update
sudo apt-get install build-essential python-pip python-dev python-smbus git
sudo pip install Adafruit-GPIO`

Of course i was used newest installation method with "pip3":

`sudo apt-get update
sudo apt-get install build-essential python3-dev python3-pip -y
sudo pip3 install Adafruit_BBIO`

and then i was try to use my simple program and... issue. 

Normally the library is copied to the directory where the application you want to run is and which references that application.
For example if i have file of my application in folder like this:
`/my_python/servotest.py`
in this same folder i must have MotorBridge.py library, like this example:
`/my_python/MotorBridge.py`

In addition, below I present the entire listing of this library for better analysis:
`# /*    
 # * MotorBridge.py 
 # * This is a library for BBG/BBB motor bridge cape 
 # *   
 # * Copyright (c) 2015 seeed technology inc.  
 # * Author      : Jiankai Li  
 # * Create Time:  Nov 2015
 # * Change Log : 
 # *
 # * The MIT License (MIT)
 # *
 # * Permission is hereby granted, free of charge, to any person obtaining a copy
 # * of this software and associated documentation files (the "Software"), to deal
 # * in the Software without restriction, including without limitation the rights
 # * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 # * copies of the Software, and to permit persons to whom the Software is
 # * furnished to do so, subject to the following conditions:
 # * 
 # * The above copyright notice and this permission notice shall be included in
 # * all copies or substantial portions of the Software.
 # * 
 # * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 # * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 # * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 # * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 # * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 # * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 # * THE SOFTWARE.
 # */
# FileName : MotorBridge.py
# by Jiankai.li

import Adafruit_GPIO.I2C as I2C
import Adafruit_BBIO.GPIO as GPIO
import time
Reset = "P9_23"
MotorBridge = I2C.Device(0x4b, 2)
GPIO.setup(Reset, GPIO.OUT)

ReadMode  = 0
WriteMode = 1
[... 351 lines cut ...]
            WriteHalfWord(SVM5_ANGLE,Angle) 
            time.sleep(DelayTime)

        if ServoName == 6:
            WriteHalfWord(SVM6_ANGLE,Angle)   
            time.sleep(DelayTime)    
            
def myloop():
    print ('Hello From MotorBridge')
    time.sleep(1)
    motor.StepperMotorBMove(-1000,1000) # 20 steppers  1000us every step
    time.sleep(1)
    motor.StepperMotorBMove(1000,1000)  # 20 steppers  1000us every step
    myloop()

if __name__=="__main__":
    
    motor = MotorBridgeCape()
    motor.StepperMotorBInit()
    motor.StepperMotorBMove(1000,1000) # 20 steppers  1000us every step
    myloop()
`

Forgive my long description but I don't speak English well.
Thank you for your response and help. Maybe you can figure out what this topic is about.

tomek242, 2021-02-19:
Anyone have some idea?

pdp7, 2021-02-20:
@tomek242 sorry for the delay.  Please try running with strace so we might be able to see where the error occurs:
`sudo strace -o /tmp/strace-log.txt -f python test.py` and change test.py to be whatever your script is.  And then upload strace-log.txt here

This issues in the MotoBridgeCape might also be related:
https://github.com/Seeed-Studio/MotorBridgeCapeforBBG_BBB/issues/6

@Pillar1989 do you know who at Seeed is working on the BBG_MotorBridgeCape?  Thank you

silver2row, 2021-02-27:
Hello,

It is late now but I have some resources on this effort...

https://github.com/silver2row/bbg is the link I created some time ago. I keep updating it but sometimes I shy away from it all at once.

Seth

P.S. If you would like to have it work, just follow that repo. and if you have any issues, just ask away.

tomek242, 2021-02-28:
Hello.
I just had more time to test everything and thats the results of `sudo strace -o /tmp/strace-log.txt -f python servotest.py`

`Traceback (most recent call last):
  File "servotest.py", line 1, in <module>
    import MotorBridge
  File "/var/lib/cloud9/my_python/MotorBridge.py", line 33, in <module>
    import Adafruit_GPIO.I2C as I2C
ImportError: No module named Adafruit_GPIO.I2C`

In other side i was tested MotorBridge.py:
`Traceback (most recent call last):
  File "MotorBridge.py", line 33, in <module>
    import Adafruit_GPIO.I2C as I2C
ImportError: No module named Adafruit_GPIO.I2C`

... same problem `No module named Adafruit_GPIO.I2C`

And this is the strace log:
2268  execve("/usr/bin/python", ["python", "MotorBridge.py"], 0xbe958790 /* 14 vars */) = 0
2268  brk(NULL)                         = 0x1d98000
2268  uname({sysname="Linux", nodename="beaglebone", ...}) = 0
2268  mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f30000
2268  access("/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
2268  openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
2268  fstat64(3, {st_mode=S_IFREG|0644, st_size=128526, ...}) = 0
2268  mmap2(NULL, 128526, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6eea000
2268  close(3)                          = 0
2268  openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libpthread.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
2268  read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0001M\0\0004\0\0\0"..., 512) = 512
2268  _llseek(3, 108864, [108864], SEEK_SET) = 0
2268  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1400) = 1400
2268  _llseek(3, 74184, [74184], SEEK_SET) = 0
2268  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
2268  _llseek(3, 108864, [108864], SEEK_SET) = 0
2268  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1400) = 1400
2268  _llseek(3, 74184, [74184], SEEK_SET) = 0
2268  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
2268  _llseek(3, 108864, [108864], SEEK_SET) = 0
2268  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1400) = 1400
2268  _llseek(3, 74184, [74184], SEEK_SET) = 0
2268  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
2268  _llseek(3, 108864, [108864], SEEK_SET) = 0
2268  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1400) = 1400
2268  _llseek(3, 74184, [74184], SEEK_SET) = 0
2268  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
2268  _llseek(3, 108864, [108864], SEEK_SET) = 0
2268  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1400) = 1400
2268  _llseek(3, 74184, [74184], SEEK_SET) = 0
2268  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
2268  _llseek(3, 108864, [108864], SEEK_SET) = 0
2268  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1400) = 1400
2268  _llseek(3, 74184, [74184], SEEK_SET) = 0
2268  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
2268  fstat64(3, {st_mode=S_IFREG|0755, st_size=110264, ...}) = 0
2268  mmap2(NULL, 148080, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ec5000
2268  mprotect(0xb6ed7000, 61440, PROT_NONE) = 0
2268  mmap2(0xb6ee6000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x11000) = 0xb6ee6000
2268  mmap2(0xb6ee8000, 4720, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6ee8000
2268  close(3)                          = 0
2268  openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
2268  read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0000\n\0\0004\0\0\0"..., 512) = 512
2268  _llseek(3, 8692, [8692], SEEK_SET) = 0
2268  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
2268  _llseek(3, 8340, [8340], SEEK_SET) = 0
2268  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
2268  _llseek(3, 8692, [8692], SEEK_SET) = 0
2268  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
2268  _llseek(3, 8340, [8340], SEEK_SET) = 0
2268  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
[... 844 lines cut ...]
2268  openat(AT_FDCWD, "/usr/lib/python2.7/dist-packages/Adafruit_GPIOmodule.so", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
2268  openat(AT_FDCWD, "/usr/lib/python2.7/dist-packages/Adafruit_GPIO.py", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
2268  openat(AT_FDCWD, "/usr/lib/python2.7/dist-packages/Adafruit_GPIO.pyc", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
2268  stat64("/usr/lib/python2.7/dist-packages/gtk-2.0/Adafruit_GPIO", 0xbee00230) = -1 ENOENT (No such file or directory)
2268  openat(AT_FDCWD, "/usr/lib/python2.7/dist-packages/gtk-2.0/Adafruit_GPIO.arm-linux-gnueabihf.so", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
2268  openat(AT_FDCWD, "/usr/lib/python2.7/dist-packages/gtk-2.0/Adafruit_GPIO.so", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
2268  openat(AT_FDCWD, "/usr/lib/python2.7/dist-packages/gtk-2.0/Adafruit_GPIOmodule.so", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
[... cut ...]
2268  write(2, "ImportError", 11)       = 11
2268  write(2, ": ", 2)                 = 2
2268  write(2, "No module named Adafruit_GPIO.I2"..., 33) = 33
2268  write(2, "\n", 1)                 = 1
2268  rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6d2bca1}, {sa_handler=0x5b8e0d, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6d2bca1}, 8) = 0
2268  munmap(0xb674c000, 262144)        = 0
2268  exit_group(1)                     = ?
2268  +++ exited with 1 +++

tomek242, 2021-02-28:
> Hello,
>
> It is late now but I have some resources on this effort...
>
> https://github.com/silver2row/bbg is the link I created some time ago. I keep updating it but sometimes I shy away from it all at [...]

Hey Seth I don't saw that line `filepath = "/dev/i2c-2".format(bus)` in MotorBridge.py file.

tomek242, 2021-02-28:
> Hello,
>
> It is late now but I have some resources on this effort...
>
> https://github.com/silver2row/bbg is the link I created some time ago. I keep updating it but sometimes I shy away from it all at [...]

Hello Seth. Where to find smbus2.py to edit this 302 line?

silver2row, 2021-03-01:
Hello @tomek242 ,

Seth here. Okay, so `pip3 install smbus2` is first. And...you may want to create a python3 venv w/ `sudo apt install python3-venv` and then `python3 -m venv <some_directory>` .

...

Then, go into the Python venv by typing source bin/activate. I am sure you are already familiar w/ these commands. You may also want to track things w/ a `requirements.txt` file. After you create that file, type `pip3 freeze > requirements.txt` .

Then...to find that line 302 in `/smbus/smbus.py`, look in this directory: `/lib/python3.7/site-packages/smbus2/smbus2.py` .

Also...in the file pyvenv.cfg, exchange out the false w/ true.

Seth

P.S. If you need any additional ideas, please just reply.

silver2row, 2021-03-01:
Hello @tomek242 ,

If that is not clear or if you need additional help, just reply. So, set up the venv, go into the venv w/ souce bin/activate, install smbus2 w/ pip3, and then go to the listed dir. at /lib/python3.7/site-packages/smbus2/ while in the venv directly in the venv.

smbus2.py will be located in that dir.

Seth

P.S. Also, stay in the venv to run the file.py of your choice w/ the Motorbridge.py altered file. If you need the altered file, reply and I will do my best to update you.

tomek242, 2021-03-01:
Hello guys and i have this:
`filepath = "/dev/i2c-{}".format(bus)`
and a should have this:
`filepath = "/dev/i2c-2".format(bus)`
sorry for maybe stupid question but it is diffirence between this two codes, the two parentheses ?

silver2row, 2021-03-01:
@tomek242 ,

Hello...

    pip3 freeze > requirements.txt

This makes it so you can have all your 'birds in one dish.' I mean, you can have those virtual or venv w/ all the python3 libs.

    filepath = "/dev/i2c-2".format(bus)

This makes it so you can use i2c-2 device on the BBB instead of what random guess the software makes to pick a randomly generated i2c-<device>.

You did not change the MotorBridge.py file yet. Look here: https://github.com/silver2row/bbg/blob/master/MBC/MotorBridge.py 

See that difference and see what your MotorBridge.py file looks like?

Check out mine and let me know what you think. 

Seth

P.S. Remember, we have to get MotorBridge.py working first. Then, we can move on to your source.

silver2row, 2021-03-04:
@tomek242 ,

Hello...Seth here. Did you get it up and running yet? You may want to update/upgrade your packages and kernel to --lts-4_19. There are some updates in this kernel too w/ it being r59.

Seth

P.S. Please reply when you have time...

tomek242, 2021-03-21:
Hello Seth, hello silver2row.
Im sorry for late answer but i had many personal issues since lasy month - not only with my Beagle :-(
Anyway, it seems to be install all right. Lines in my file smbus2.py looks like below:

`302"""
303        if isinstance(bus, int):
304            filepath = "/dev/i2c-2".format(bus) **#line was changed as u suggest**
305        elif isinstance(bus, str):
306            filepath = bus
307        else:
308            raise TypeError("Unexpected type(bus)={}".format(type(bus)))
308
309        self.fd = os.open(filepath, os.O_RDWR)
310        self.funcs = self._get_funcs()`

I made another attempt to run my servo test program. Results below:

`import MotorBridge
import time
ServoName   =  1
Frequency   =  50
Angle1      =  20
Angle2      =  160
    
if __name__=="__main__":
    motor = MotorBridge.MotorBridgeCape()
    motor.ServoInit(ServoName,Frequency)
    
    while True:
        print ('Servo Test')
        motor.ServoMoveAngle(ServoName,Angle1)
        time.sleep(2)
        motor.ServoMoveAngle(ServoName,Angle2)
        time.sleep(2)`

Trying run and:

`
Traceback (most recent call last):
  File "/var/lib/cloud9/my_python/servo.py", line 11, in <module>
    motor = MotorBridge.MotorBridgeCape()
  File "/var/lib/cloud9/my_python/MotorBridge.py", line 229, in __init__
    GPIO.output(Reset, GPIO.HIGH)
NameError: name 'GPIO' is not defined
`
May be i don't understeand anything or im too stupid to make this... everything seems work correctly bot it isn't:
 Below i put line229 from MotorBridge.py:
`
227 class MotorBridgeCape:
228     def __init__(self):
229         GPIO.output(Reset, GPIO.HIGH) `# ....and what is wrong in here?`
230         time.sleep(1)
` 

Greetings guys and thanks for ur answer and help....

silver2row, 2021-03-23:
Hello @tomek242 ,

Did you put the BB-UART2-00A0.dtbo in the /boot/uEnv.txt file or use config-pin to handle the UART2 pins?

This may be the cause. Oh and try this source at where you typed:

    if name=="main":

But instead of that line, use this line:

    if __name__=="__main__":

Also...

Please reply w/ this command `cat /boot/uEnv.txt`.

Seth

silver2row, 2021-03-23:
Hello @tomek242 ,

Also, the pins on the Motor Bridge Cape are confusing for the servo header. If you are using Servo1 in your source, try to use the exterior pin(s) on the servo header. So, you may be plugged into the sixth servo pin instead of the first servo pin. Try to alter your configuration and...

Perform the necessary adjustments and please reply.

Seth

silver2row, 2021-04-14:
@tomek242 ,

Hello sir...did you figure it out yet? I am checking in to see if you got things working.

Seth

tomek242, 2021-05-20:
Hello Seth. Im sorry for late like always. BBB its my hobby only so i can write about it only when i had time after work.
Anyway unfortunatelly i had no result - im using of course new wersion of  I2C library, i had exactly same start of this file:

`from smbus2 import SMBus
import time
import pathlib

# reset pin is P9.23, i.e. gpio1.17
reset_pin = pathlib.Path('/sys/class/gpio/gpio49/direction')
reset_pin.write_text('low')

bus = SMBus('/dev/i2c-2')

ReadMode  = 0
WriteMode = 1
DeAddr    = 0X4B
ConfigValid =  0x3a6fb67c
DelayTime = 0.00`

also i've checked couple times lines since line 201 and rest, and i have exactly same code:

`def WriteByte(Reg,Value):
    data = [0 for i in range(2)]
    data[0] = Reg
    data[1] = Value
    bus.write_i2c_block_data(0x4b, 1, data)

def WriteHalfWord(Reg,Value):
    data = [0 for i in range(3)]
    data[0] = Reg
    data[1] = Value & 0xff
    data[2] = (Value>>8) & 0xff
    bus.write_i2c_block_data(0x4b, 1, data)

def WriteOneWord(Reg,Value):
    data = [0 for i in range(5)]
    data[0] = Reg
    data[1] = Value & 0xff
    data[2] = (Value>>8) & 0xff
    data[3] = (Value>>16) & 0xff
    data[4] = (Value>>24) & 0xff
    bus.write_i2c_block_data(0x4b, 1, data)

def SetDefault():
    WriteOneWord(CONFIG_VALID, 0x00000000)

class MotorBridgeCape:
    def __init__(self):
        GPIO.output(Reset, GPIO.HIGH)
        time.sleep(1)`

So everything seems correct but i had exactly same error like before in my posts....

silver2row, 2021-05-31:
Hello @tomek242 ,

Okay...I will test again. I have the hardware here. I will let you know exactly my set up, source, and board used. Please be patient.

Seth

silver2row, 2021-05-31:
Okay and okay,

@tomek242 :

Here:

```
 # * MotorBridge.py
 # * This is a library for BBG/BBB motor bridge cape
 # * Copyright (c) 2015 seeed technology inc.
 # * Author      : Jiankai Li
 # * Create Time:  Nov 2015
 # * Change Log : Jan. 2020 by Seth w/ help from #beagle on Freenode
 # *
 # * The MIT License (MIT)
 # *
 # * Permission is hereby granted, free of charge, to any person obtaining a copy
 # * of this software and associated documentation files (the "Software"), to deal
 # * in the Software without restriction, including without limitation the rights
 # * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 # * copies of the Software, and to permit persons to whom the Software is
 # * furnished to do so, subject to the following conditions:
 # *
 # * The above copyright notice and this permission notice shall be included in
 # * all copies or substantial portions of the Software.
 # *
 # * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 # * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 # * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 # * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 # * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 # * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 # * THE SOFTWARE.
 # */
# FileName : MotorBridge.py
# by Jiankai.li

from smbus2 import SMBus
import time
import pathlib

# reset pin is P9.23, i.e. gpio1.17
reset_pin = pathlib.Path('/sys/class/gpio/gpio49/direction')
reset_pin.write_text('low')

bus = SMBus('/dev/i2c-2')

ReadMode  = 0
WriteMode = 1
DeAddr    = 0X4B
ConfigValid =  0x3a6fb67c
DelayTime = 0.005

# TB_WORKMODE

TB_SHORT_BREAK  = 0
TB_CW           = 1
TB_CCW          = 2
TB_STOP         = 3
TB_WORKMODE_NUM = 4

# TB_PORTMODE

TB_DCM          = 0
TB_SPM          = 1
TB_PORTMODE_NUM = 2

# SVM_PORT

SVM1            = 0
[... 381 lines cut ...]
        motor.DCMotorMove(2, 2, 90)

#All Flanks...Straight Ahead!
    for i in range(1, 51):
        motor.DCMotorMove(1, 1, 90)
        motor.DCMotorMove(2, 1, 90)

#Make another U-Turn but this time, motors go in the opposite direction!
    for i in range(1, 3):
        motor.DCMotorMove(1, 2, 90)
        motor.DCMotorMove(2, 1, 90)

#Straight Ahead
    for i in range(1, 51):
        motor.DCMotorMove(1, 1, 90)
        motor.DCMotorMove(2, 1, 90)

    print("Testing, Testing\n, 1\n, 2\n, 3\n")
```

Then, after those files are in your cloned git repo. from https://github.com/silver2row/bbg and after changing line 302 to what this specific repo. states for smbus2, add the BB-UART2-00A0.dtbo in your /boot/uEnv.txt file under the uboot overlays section.

Seth

P.S. I think this covers it. Please let me know what you think thus far. I am not sure what I could be missing since mine works.

silver2row, 2021-05-31:
@tomek242 ,

Oh and...

If you have time, please use a venv for this specific instance if you are having issues. So, `sudo apt install python3-venv` and then type this command for your venv after cloning the repo. from above, `python3 -m venv bbg` . 

Seth

P.S. This may help you exclude specific issues so we can debug a bit better. Also, while in the venv called bbg, use `pip3 freeze > requirements.txt` . If that requirements.txt file is not already available, make it with touch requirements.txt.