adafruit/adafruit-beaglebone-io-python #349: Debugging PWM and Adafruit_BBIO on the BBBlue...

silver2row, 2021-03-21:
Hello,

I am starting to receive some issues w/ PWM on the BBBlue that I cannot figure out...

`DEBUG: Adafruit_BBIO: set_pin_mode() :: Pinmux file: /sys/devices/platform/ocp/ocp:P9_14_pinmux/state, mode: pwmDEBUG: Adafruit_BBIO: set_pin_mode(): beaglebone_blue() is TRUE; return BBIO_OK`

I am not sure if this PWM issue pertains to the BBBlue or my source...

Seth

P.S. I thought I could figure it out but I see no headings in research w/ this issue.

silver2row, 2021-03-21:
Hello,

Here is the required info. from version.sh.

```
# This is sudo /opt/scripts/tools/version.sh
 
git:/opt/scripts/:[23def497e973fd294680f671af6b6549f4dadb65]
eeprom:[A335BNLTBLA21708EL000258]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600)]:[location: dd MBR]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblue.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.19.94-ti-r60]
nodejs:[v10.24.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210225.0-0~buster+20210225]
pkg:[bb-customizations]:[1.20210225.0-0~buster+20210225]
pkg:[bb-usb-gadgets]:[1.20200504.0-0~buster+20200504]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin s                                                                                    pi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1                                                                                    990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   11.137685] remoteproc remoteproc0: 4a334000.pru is available
[   11.144657] remoteproc remoteproc1: 4a338000.pru is available
[   55.965562] remoteproc remoteproc2: wkup_m3 is available
[   55.976862] remoteproc remoteproc2: powering up wkup_m3
[   55.976886] remoteproc remoteproc2: Booting fw image am335x-pm-firmware.elf, size 217168
[   55.977138] remoteproc remoteproc2: remote processor wkup_m3 is now up
dmesg | grep pru
[   11.137685] remoteproc remoteproc0: 4a334000.pru is available
[   11.137869] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   11.144657] remoteproc remoteproc1: 4a338000.pru is available
[   11.144846] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.914453] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

Seth

P.S. Please let me know what you find (if anything). Sorry.

pdp7, 2021-03-21:
This think this might be related to https://github.com/adafruit/adafruit-beaglebone-io-python/issues/313

I'll need to see if I can find my Blue so I can try to test.

silver2row, 2021-03-21:
Hello Sir,

Yes sir...I will stay patient. I know everyone is working on new projects, i.e. esp. w/ the BeagleV coming out. So, I will stay patient in case you reply.

Seth

P.S. I looked over #313 .

silver2row, 2021-03-22:
Hello Sir,

The udev rules were commented out for pwm to be in /dev/pwm/. Anyway...I will close this issue now. I am still working on the lib. for PWM.

Seth

closed 2021-03-22 by silver2row