adafruit/adafruit-beaglebone-io-python #352: Set pin mode failed for uart channel. (UART4)

zach323, 2021-06-17:
Hi there, I am trying to use Adafruit UART to enable UART4 to work with some modbus devices, but I am getting an error if I try to setup("UART4"). 
**code**
`import Adafruit_BBIO.UART as u`
`u.setup("UART4")`

`py_setup_uart(UART4): set_pin_mode() failed for pin=P9_11Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Set pin mode failed for uart channel.
`

****sudo /opt/scripts/tools/version.sh output:****
```

git:/opt/scripts/:[a252576b47d48e1e770ac258eb25e831bfd8c580]
eeprom:[A335BNLTEIA02034SBB06985]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2019.04-00002-gc9b3922522 (Aug 24 2020 - 16:42:18 -0500)]:[location: dd MBR]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2019.04-00002-g31a8ae0206 (May 13 2020 - 09:26:17 -0500)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g31a8ae0206]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BBORG_COMMS-00A2]
UBOOT: Loaded Overlay:[BB-UART4-00A0]
kernel:[4.14.108-ti-r143]
nodejs:[v6.17.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-UART1-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210416.0-0~stretch+20210416]
pkg:[bb-customizations]:[1.20210225.0-0~stretch+20210225]
WARNING:pkg:[bb-usb-gadgets]:[NOT_INSTALLED]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~stretch+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BBORG_COMMS, coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.147401] remoteproc remoteproc0: wkup_m3 is available
[    1.231470] remoteproc remoteproc0: powering up wkup_m3
[    1.231593] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[    1.234048] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.789145] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.802115] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

```
Here is my uEnv.txt. I have seen a lot of inconsistent information on uboot overlays, whether you have to use them, etc, what to change etc. Lots of google digging and conflicting information depending on the year.

```#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.108-ti-r143
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_rproc (4.19.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti, 4.19.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```

I noticed that the P9.11 and P9.13 pins for UART4 aren't even showing up in my 'ls /sys/devices/platform/ocp | grep pinmux*'

```
ls /sys/devices/platform/ocp/ | grep pinmux*
ocp:A15_pinmux
ocp:P8_07_pinmux
ocp:P8_08_pinmux
ocp:P8_09_pinmux
ocp:P8_10_pinmux
ocp:P8_11_pinmux
ocp:P8_12_pinmux
ocp:P8_13_pinmux
ocp:P8_14_pinmux
ocp:P8_15_pinmux
ocp:P8_16_pinmux
ocp:P8_17_pinmux
ocp:P8_18_pinmux
ocp:P8_19_pinmux
ocp:P8_26_pinmux
ocp:P8_27_pinmux
ocp:P8_28_pinmux
ocp:P8_29_pinmux
ocp:P8_30_pinmux
ocp:P8_31_pinmux
ocp:P8_32_pinmux
ocp:P8_33_pinmux
ocp:P8_34_pinmux
ocp:P8_35_pinmux
ocp:P8_36_pinmux
ocp:P8_37_pinmux
ocp:P8_38_pinmux
ocp:P8_39_pinmux
ocp:P8_40_pinmux
ocp:P8_41_pinmux
ocp:P8_42_pinmux
ocp:P8_43_pinmux
ocp:P8_44_pinmux
ocp:P8_45_pinmux
ocp:P8_46_pinmux
ocp:P9_12_pinmux
ocp:P9_14_pinmux
ocp:P9_15_pinmux
ocp:P9_16_pinmux
ocp:P9_17_pinmux
ocp:P9_18_pinmux
ocp:P9_19_pinmux
ocp:P9_20_pinmux
ocp:P9_21_pinmux
ocp:P9_22_pinmux
ocp:P9_23_pinmux
ocp:P9_25_pinmux
ocp:P9_27_pinmux
ocp:P9_28_pinmux
ocp:P9_29_pinmux
ocp:P9_30_pinmux
ocp:P9_31_pinmux
ocp:P9_41_pinmux
ocp:P9_42_pinmux
ocp:P9_91_pinmux
ocp:P9_92_pinmux
```
I am still learning a lot about this device, but any help you could provide would be great. Thanks.

RobertCNelson, 2021-06-17:
```
UBOOT: Loaded Overlay:[BBORG_COMMS-00A2]
UBOOT: Loaded Overlay:[BB-UART4-00A0]
```
These two overlays got loaded for uart features. Thus if they used the pin, they would delete the universal "pin mode" option from config-pin and just enable usart on bootup..

Regards,

zach323, 2021-06-17:
> ```
> UBOOT: Loaded Overlay:[BBORG_COMMS-00A2]
> UBOOT: Loaded Overlay:[BB-UART4-00A0]
> ```
> 
> These two overlays got loaded for uart features. Thus if they used the pin, they would delete the universal "pin mode" option from config-pin and just enable usart on bootup..
> 
> https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-UART4-00A0.dts#L36-L54
> 
> @pdp7 uart is already enabled, does this library have way to just use uart avaialble?
> 
> Regards,

1) How do you disable BBORG_COMMS_00A2 from loading?
2) Does Adafruit need overlays loaded to use UART?

Thanks for the help!

RobertCNelson, 2021-06-17:
HI @zach323 since the uart is already enabled on bootup, you should be able to access /dev/ttyS* thru python without the help of this io library.

Regards,

silver2row, 2021-08-07:
Hello,

So...just for sake of argument here, would one just use pyserial instead of ` UART.setup("P9_xx") ` ?

Seth