adafruit/adafruit-beaglebone-io-python #353: Cannot initialize GPIO.INPUT on BeagleBone Black Ubuntu 20.04

mrunaljsarvaiya, 2021-07-21:
I'm running into using when initializing a GPIO pin as input on a BeagleBone Black running Ubuntu 20.04 (https://forum.digikey.com/t/ubuntu-server-20-04/6649/25). Here is the python3 code

```
#! /usr/bin/env python3

import time
import Adafruit_BBIO.GPIO as GPIO
import Adafruit_BBIO.PWM as PWM
import Adafruit_BBIO.ADC as ADC
from Adafruit_I2C import Adafruit_I2C

def test_GPIO_output():
    channel = "P8_10"
    GPIO.setup(channel, GPIO.IN)
    GPIO.cleanup()

if __name__ == '__main__':
    test_GPIO_output()
```
When I run the script above I get the following error
```
Traceback (most recent call last):
  File "test.py", line 15, in <module>
    test_GPIO_output()
  File "test.py", line 11, in test_GPIO_output
    GPIO.setup(channel, GPIO.IN)
ValueError: Set gpio mode failed, missing file or invalid permissions.

```

Here is the output of my system to  ```sudo /opt/scripts/tools/version.sh ```

```
git:/opt/scripts/:[4d3a0aff57aae78a0db890f72ce505f61f1a93ce]
eeprom:[A335BNLTEIA02033SBB11671]
model:[TI_AM335x_BeagleBone_Black]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-dirty]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g31a8ae0206]:[location: dd MBR]
kernel:[5.4.0-26-generic]
/boot/uEnv.txt Settings:
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200430.0-0~focal+20200430]
WARNING:pkg:[bb-wl18xx-firmware]:[NOT_INSTALLED]
pkg:[kmod]:[27-1ubuntu2]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input netdev i2c bluetooth gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait]
dmesg | grep remote
[  105.467106] remoteproc remoteproc0: wkup_m3 is available
dmesg | grep pru
dmesg | grep pinctrl-single
[    2.826750] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

```

RaiBnod, 2022-04-07:
**Same issue on:** [AM3358 Debian 10.3 2020-04-06 4GB SD IoT](https://debian.beagleboard.org/images/bone-debian-10.3-iot-armhf-2020-04-06-4gb.img.xz)

**Code:** `GPIO.setup("P9_31", GPIO.IN)`

**Exception:** ValueError: Set gpio mode failed, missing file or invalid permissions.

RobertCNelson, 2022-04-07:
P9_31 = audio... you have hdmi audio disabled right?

Regards,