adafruit/adafruit-beaglebone-io-python #356: Can't Seem to Install Adafruit BBIO Library

keenanjohnson, 2021-11-08:
Hello I'm trying to install the Adafruit_BBIO library on a BeagleBone Black, but I'm having a weird error when I try to do so that I can't seem to get around. I'm trying to install the library because I'm trying to use the Adafruit Blinka library on the Raspberry Pi. When I run my python file that uses the Blinka LIbrary, I get the following error telling me to install the BBIO library. Any help would be greatly appreciated! I've added all of the logs and information below that I can! 

Thanks!

**Blinka Error**
```
root@d530b65:/usr/src# python co2.py 
Traceback (most recent call last):
  File "/usr/src/co2.py", line 24, in <module>
    import board
  File "/usr/local/lib/python3.10/site-packages/board.py", line 79, in <module>
    from adafruit_blinka.board.beagleboard.beaglebone_black import *
  File "/usr/local/lib/python3.10/site-packages/adafruit_blinka/board/beagleboard/beaglebone_black.py", line 2, in <module>
    from adafruit_blinka.microcontroller.am335x import pin
  File "/usr/local/lib/python3.10/site-packages/adafruit_blinka/microcontroller/am335x/pin.py", line 2, in <module>
    import Adafruit_BBIO.GPIO as GPIO
ModuleNotFoundError: No module named 'Adafruit_BBIO'
```

**System Infomation**

Beaglebone Black running [Balena](https://www.balena.io/)
Here is the Dockerfile that configures the container I'm trying to install Adafruit_BBIO into:

```
FROM balenalib/%%BALENA_MACHINE_NAME%%-python

RUN install_packages vim build-essential wget gpsd-clients dbus

# Install Python Packages
RUN pip install smbus adafruit-circuitpython-scd30 influxdb-client paho-mqtt requests gpsd-py3 adafruit-circuitpython-ina219 adafruit-circuitpython-lc709203f adafruit-circuitpython-dps310
RUN pip install numpy --upgrade

RUN pip install Adafruit_BBIO

WORKDIR /usr/src/
COPY . ./

CMD sleep infinity
```

**Library Installation Error Output:**
```
[co2]           Collecting Adafruit_BBIO
[co2]             Downloading Adafruit_BBIO-1.2.0.tar.gz (78 kB)
[co2]           Using legacy 'setup.py install' for Adafruit-BBIO, since package 'wheel' is not installed.
[co2]           Installing collected packages: Adafruit-BBIO
[co2]               Running setup.py install for Adafruit-BBIO: started
[co2]               Running setup.py install for Adafruit-BBIO: finished with status 'error'
[co2]               ERROR: Command errored out with exit status 1:
[co2]                command: /usr/local/bin/python3.10 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/setup.py'"'"'; __file__='"'"'/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-4vehuikb/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.10/Adafruit-BBIO
[co2]                    cwd: /tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/
[co2]               Complete output (54 lines):
[co2]               running install
[co2]               running build
[co2]               running build_py
[co2]               creating build
[co2]               creating build/lib.linux-armv7l-3.10
[co2]               copying Adafruit_I2C.py -> build/lib.linux-armv7l-3.10
[co2]               creating build/lib.linux-armv7l-3.10/overlays
[co2]               copying overlays/__init__.py -> build/lib.linux-armv7l-3.10/overlays
[co2]               copying overlays/builder.py -> build/lib.linux-armv7l-3.10/overlays
[co2]               creating build/lib.linux-armv7l-3.10/Adafruit_BBIO
[co2]               copying Adafruit_BBIO/sysfs.py -> build/lib.linux-armv7l-3.10/Adafruit_BBIO
[co2]               copying Adafruit_BBIO/Encoder.py -> build/lib.linux-armv7l-3.10/Adafruit_BBIO
[co2]               copying Adafruit_BBIO/__init__.py -> build/lib.linux-armv7l-3.10/Adafruit_BBIO
[co2]               running build_ext
[co2]               building 'Adafruit_BBIO.GPIO' extension
[co2]               creating build/temp.linux-armv7l-3.10
[co2]               creating build/temp.linux-armv7l-3.10/source
[... 13 lines cut ...]
[co2]                     |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
[co2]               source/py_gpio.c:603:7: warning: ‘PyEval_InitThreads’ is deprecated [-Wdeprecated-declarations]
[co2]                 603 |       PyEval_InitThreads();
[co2]                     |       ^~~~~~~~~~~~~~~~~~
[co2]               In file included from /usr/local/include/python3.10/Python.h:144,
[co2]                                from source/py_gpio.c:31:
[co2]               /usr/local/include/python3.10/ceval.h:122:37: note: declared here
[co2]                 122 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
[co2]                     |                                     ^~~~~~~~~~~~~~~~~~
[co2]               gcc -pthread -shared build/temp.linux-armv7l-3.10/source/c_pinmux.o build/temp.linux-armv7l-3.10/source/common.o build/temp.linux-armv7l-3.10/source/constants.o build/temp.linux-armv7l-3.10/source/event_gpio.o build/temp.linux-armv7l-3.10/source/py_gpio.o -L/usr/local/lib -o build/lib.linux-armv7l-3.10/Adafruit_BBIO/GPIO.cpython-310-arm-linux-gnueabihf.so
[co2]               /usr/bin/ld: build/temp.linux-armv7l-3.10/source/py_gpio.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:15: multiple definition of `version'; build/temp.linux-armv7l-3.10/source/constants.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:15: first defined here
[co2]               /usr/bin/ld: build/temp.linux-armv7l-3.10/source/py_gpio.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:14: multiple definition of `both_edge'; build/temp.linux-armv7l-3.10/source/constants.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:14: first defined here
[co2]               /usr/bin/ld: build/temp.linux-armv7l-3.10/source/py_gpio.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:13: multiple definition of `falling_edge'; build/temp.linux-armv7l-3.10/source/constants.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:13: first defined here
[co2]               /usr/bin/ld: build/temp.linux-armv7l-3.10/source/py_gpio.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:12: multiple definition of `rising_edge'; build/temp.linux-armv7l-3.10/source/constants.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:12: first defined here
[co2]               /usr/bin/ld: build/temp.linux-armv7l-3.10/source/py_gpio.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:11: multiple definition of `pud_down'; build/temp.linux-armv7l-3.10/source/constants.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:11: first defined here
[co2]               /usr/bin/ld: build/temp.linux-armv7l-3.10/source/py_gpio.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:10: multiple definition of `pud_up'; build/temp.linux-armv7l-3.10/source/constants.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:10: first defined here
[co2]               /usr/bin/ld: build/temp.linux-armv7l-3.10/source/py_gpio.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:9: multiple definition of `pud_off'; build/temp.linux-armv7l-3.10/source/constants.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:9: first defined here
[co2]               /usr/bin/ld: build/temp.linux-armv7l-3.10/source/py_gpio.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:8: multiple definition of `alt0'; build/temp.linux-armv7l-3.10/source/constants.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:8: first defined here
[... cut ...]
[co2]               /usr/bin/ld: build/temp.linux-armv7l-3.10/source/py_gpio.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:4: multiple definition of `high'; build/temp.linux-armv7l-3.10/source/constants.o:/tmp/pip-install-ld5mlyu6/adafruit-bbio_eedec34ab4e340fa9492612a3cb1a1ff/source/constants.h:4: first defined here
[co2]               collect2: error: ld returned 1 exit status
[co2]               error: command '/usr/bin/gcc' failed with exit code 1
```

djgood, 2021-11-19:
I believe this was fixed in #345. Unfortunately the code released on PyPi doesn't contain this fix yet. The workaround for us for now is to downgrade our version of Debian so that we are using an older GCC or to compile/install from master.

keenanjohnson, 2021-11-20:
Confirmed that we were able to resolve this issue in our project by downgrading to a buster release. 

See https://github.com/Ribbit-Network/ribbit-network-frog-sensor/pull/81

keenanjohnson, 2021-11-20:
I'll check if we can support pushing out a new release of the BBIO library in https://github.com/adafruit/adafruit-beaglebone-io-python/pull/345

closed 2021-11-20 by keenanjohnson