adafruit/adafruit-beaglebone-io-python #370: P9_28 - P9_29 - P9-31 pins setupa as GPIO problem

moonpars, 2024-01-17:
Debian 10.3 is installed on Beaglebone Black. I installed a new image and this library and I want to use the 3 pins I mentioned in the title as GPIO, but;
For example, when I try the command "config-pin -q P9_29":

ERROR: open() for sys/devices/platform/ocp/ocp:P9_29_pinmux/state failed, No such file or directory

I'm getting the error. I have to use these pins and I get the same error in all 3 pins. How do I convert these to GPIO? 
I'm trying to use Device Tree overaly, but I'm too amateur for this and there isn't even "/sys/devices/platform/bone_capemgr/slots", which is one of the first steps, on the system. 

There is no such file. Can you help with this?

VERSION.SH:

git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLT00C03818BBBK02DC]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
kernel:[4.19.94-ti-r42]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.0-0rcnee0~buster+20200403]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   11.217492] remoteproc remoteproc0: 4a334000.pru is available
[   11.241566] remoteproc remoteproc1: 4a338000.pru is available
[   75.301854] remoteproc remoteproc2: wkup_m3 is available
[   76.139696] remoteproc remoteproc2: powering up wkup_m3
[   76.139729] remoteproc remoteproc2: Booting fw image am335x-pm-firmware.elf, size 217168
[   76.140014] remoteproc remoteproc2: remote processor wkup_m3 is now up
dmesg | grep pru
[   11.217492] remoteproc remoteproc0: 4a334000.pru is available
[   11.217700] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   11.241566] remoteproc remoteproc1: 4a338000.pru is available
[   11.241757] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.922321] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.935663] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

THANKS&REGARDS!

RobertCNelson, 2024-01-17:
```
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
```

disable hdmi audio via: 

```
disable_uboot_overlay_audio=1
```

and you'll get acces to those pins.. https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Disable_on-board_devices

moonpars, 2024-01-19:
Thank you! It solved the problem.

Thanks&regards!

moonpars, 2024-01-19:
Thank You! It solved the problem.

Thank and regards!

closed 2024-01-19 by moonpars